rene-mt/esp8266-oled-sh1106 #3: I2C on nodeMCU

barsznica, 2016-10-12:
Hi. I tried the example on a nodeMCU 1.0 & I2C and the display was garbled.

I tried https://github.com/squix78/esp8266-oled-ssd1306, but only SH1106Brzo.h worked.
SSD1306.h, SH1106.h & SSD1306Brzo.h did not work for me (also garbled display).

I'm using 12 for SDA and 14 for SDC (SCL).

Cheers!

rene-mt, 2016-10-13:
I had similar problems when trying the original SSD1306 driver with the SH1106 display. Reason was the divergent structure of the display buffer. So are sure you are using the SH1106?

barsznica, 2016-10-16:
I can't guarantee it, but the fore-mentioned SH1106Brzo lib worked, while the  SSD1306Brzo (and other 2) did not. It is a 1.3" display. The eBay seller refers to a SSH1106.
If I was using the original SSD1306 driver, it would have worked when I plugged in a SSD1306 0.96" display.

closed 2017-02-23 by rene-mt