rene-mt/esp8266-oled-sh1106 #4: esp8266 changes need to be pulled in

tablatronix, 2017-02-22:
I had to make major changes to some stuff to get esp8266 to work.
mostly

#ifndef __SAM3X8E__ problems had to be removed or new blocks added for esp8266

Bring in updates to sd1306 lib to fix.

rene-mt, 2017-02-23:
Do you have some more details for me? I think I don't fully understand the issue :)

TonyVera, 2017-07-18:
I can't get this to work for I2C.  I changed the pin definitions:
#define OLED_SDA    D2  // Wemos D1 mini SDA pin
#define OLED_SDC    D1  // Wemos D1 mini SCL pin
The demo displays partial information in the bottom 10% of the display.  There is just fixed garbage in the top 90%.  Any suggestions?

axfatec, 2018-05-22:
Just tested with NODE-MCU  ESP 8266 and display SD1106 1.3"
using I2C  definitions:
SH1106 display(0x3c, D3, D4); 
SH1106Ui ui ( &display );
..resutls:  same issue of "TonyVera".
Could you help us ?
Thanks , Alex

tablatronix, 2018-05-22:
I will try to fork and push my changes

axfatec, 2018-05-22:
Hello.
Just to inform you that using:
 https://github.com/ThingPulse/esp8266-oled-ssd1306 

the display works correctly.  Pheraps it can help you.
regards
Alex

closed 2025-06-20 by rene-mt