rene-mt/esp8266-oled-sh1106 #5: only the even lines display

LilaiJi, 2017-04-27:
[image]
Hello Rene,
thank u at first for ur great job.
I test my 128x64 OLED (SPI) with ur demo, but I can not get my display working properly.
It uses just the even lines (2, 4, 6, ...,128), and the odd lines stay black.
Could u please help out?
Best regards

skhaz, 2017-05-05:
Remove the line `display.flipScreenVertically();`, worked for me

rene-mt, 2017-05-05:
I assume the display then might use a different driver than mine, which mainly seems to be about the correct addressing of the display buffer. Thus, I have no elaborate solution at hand. Maybe try out @skhaz's solution and tell how it went.

LilaiJi, 2017-05-05:
@skhaz Thank you! When I take the line away, it works properly.

closed 2025-06-20 by rene-mt