rene-mt/esp8266-oled-sh1106 #7: SH1106 I2C not working

hjgode, 2018-02-18:
I tested the demo on a SH1106 with I2C and it works as follows:
There is a very fast left scrolling text in line 1
the image drawn before (another demo) is still displayed (the display did no clear)

NodeMCU with SDA=D2 and SCL=D1

Lumics, 2020-08-18:
I can confirm, same issue

NeilvdM, 2020-10-10:
Has anyone ever resolved this issue with the SH1106 library working with ESP8266?
I have been stuck on this

hjgode, 2020-10-10:
For me this Library works fine with ESP-8266, see http://www.hjgode.de/wp/tag/fhem/ and https://github.com/hjgode/SH1106_mqtt_display

Therealjosephchrzempiec, 2021-01-12:
Hello i have this same problem. My skills are in hardware mostly. I'm in need of some help please. Can someone please help me. I also get the same junk links and scrolling at the bottom.

Joseph